bruhautomation/ESP-MQTT-JSON-Digital-LEDs #109: Problem with Home Assistant

Abgehoben, 2019-01-03:
Hello Guys,
At the beginn, sorry for my bad english.

I also startet with this project and have some Problems with Home Assistant i guess,
My Home Assistant start normal with the basic standart configuration.yaml, 
but when I copy the one from the Project (with the updatet one) the Home Assistant dont start or show up.
I install Home Assistant on Windows and my MQTT is from cloud.mqtt.
can some one help me pls? 
I dont know if you need more informations or not.
I say Thank you ^^

Rawet, 2019-01-05:
Hi.

I recently found out that Home Assistant have changed how to setup MQTT..

This is how you fix it:

```
light:
  - platform: mqtt
    schema: json
```

This is the old way:
```
light:
  - platform: mqtt_json
```

If this doesn't fix your problem you have probably an issue with your config, doublecheck your config with an YAML validator and post your config here.

Abgehoben, 2019-01-07:
Hello Thank you for your Help,
Already had the config.
And a Yaml validator say its all ok?
Thanks in Advance.

--- 
automation: 
  - 
    action: 
      - 
        data_template: 
          payload: "{\"transition\":{{ trigger.to_state.state | int }}}"
          topic: bruh/porch/set
        service: mqtt.publish
    alias: "Porch Animation Speed"
    hide_entity: false
    initial_state: true
    trigger: 
      - 
        entity_id: input_number.porch_animation_speed
        platform: state
input_number: 
  porch_animation_speed: 
    initial: 150
    max: 150
    min: 1
    name: "Porch Animation Speed"
    step: 10
light: 
  - 
    brightness: true
    command_topic: bruh/porch/set
    effect: true
    effect_list: 
      - bpm
      - "candy cane"
      - confetti
      - "cyclon rainbow"
      - dots
      - fire
      - glitter
      - juggle
      - lightning
      - noise
      - "police all"
      - "police one"
      - rainbow
      - "rainbow with glitter"
      - ripple
      - sinelon
      - solid
      - twinkle
    flash: true
    name: "Porch Strip"
    optimistic: false
    platform: mqtt
    qos: 0
    rgb: true
    schema: json
    state_topic: bruh/porch
mqtt: 
  broker: ym21.cloudmqtt.com
  client_id: home-assistant-1
  password: 123456xd
  port: 10024
  username: home